kercre123/wire-pod #322: [Question] Changing WiFi Network

CodaCM, 2024-04-30:
Hello, I was wondering if I could change the vectors wifi network (as I plan to move him around from time to time with the server) without having to completely reset him just to change it. I didn't see the option to do so and I'm guessing that it is not possible to do so.

Zomboy4313, 2024-04-30:
I think if you go directly to https://wpsetup.keriganc.com/ you can change Vector's WiFi there and he will keep his memory

ocueye2, 2024-06-15:
I guess if they both have the same subnet mask (the first 3 numbers of the ip like **192.168.1**.1) and you have the network name and password be the same and you have them both set to the same static address on the network and copy the state of one server to the other. it would work

CodaCM, 2024-10-30:
When the Pi image for wire-pod rolls out, do you think that would work for changing Wi-Fi? Vector only needs to connect to the Pi, and the Pi will connect to whatever network you want.