kercre123/wire-pod #323: Feature Request: Add support for custom OpenAI URL to utilize third-party OpenAI API services

BUJIDAOVS, 2024-05-04:
I have observed that when integrating with GPT, it defaults to accessing api.openai.com. However, I would like to make API requests from a different URL or proxy url. I kindly request the addition of this functionality.

dm94, 2024-05-15:
This can be very useful as many local systems have created an API similar to OpenAI so that local models such as llama3 could be used.

NonaSuomy, 2024-06-06:
I attempted to add this a while ago the issue wire told me is that most models are too wordy for vector to not timeout.

I got it to sort of work with textgen UI by setting the model prompt up with "Your answers have to be 255 characters or less" then forcing the settings into vectors backend to use the basic open API but with the local IP address. It would still go bonkers. Vector would make the WORP sound and fail.

green-codes, 2024-06-06:
Strongly second this, it would make so much sense for the self-hosted wire-pod to also use a self-hosted LM, or at least make it an option.

green-codes, 2025-06-08:
> Any news on this? I would really like to use my openai provider with my vector

I believe this is already possible by using the custom provider in the knowledge graph settings. Don't know why the issue isn't closed yet, I might be wrong?