kercre123/wire-pod #325: [HELP] https://wpsetup.keriganc.com/ can't be reached

kengengler, 2024-05-04:
I'm following the directions to get wire-pod working, but when I go to https://wpsetup.keriganc.com/ in Chrome, I get the message that the site can't be reached.

I'm trying to run wire-pod on Ubuntu/x86. And I have a prod/regular bot.

Also, as a follow-on, I did manage to get a bit farther using https://vector.techshop82.com/html/main.html, but when I reach the part where I enter the wifi, the web page says its downloading, I see vector show the animated wifi display, then the cloud with spinner, then a cloud with a exclamation point (which I assume is bad). Not really sure what's going on at that point.

I should also mention that using the https://vector.techshop82.com/html/main.html, I did manage to get it to the point where I could proceed to the wire-pod step of "Bot Setup". But when I'd enter the wifi said/password, I'd get a message back saying the password was incorrect (which it isn't). And vector just displays an animated wifi signal. Also, its a 2.4GHz wireless - not even a 5G under the same name.
[image]

kengengler, 2024-05-04:
Just to add a little more info. Using the "wire pod interface" page from my Ubuntu server, I was able to pair to my Vector again. And it seems to jump immediately to "Starting OTA update...". But, maybe because it's not really altogether ready, I see the Vector display showing a cloud with spinning and periodically, the exclamation point. And in the logs on the server, I see this repeatedly:

kercre123, 2024-05-04:
wpsetup.keriganc.com should be back up now.