kercre123/wire-pod #326: [HELP] sudo ./update.sh

choozuradventure, 2024-05-05:
Original installation on a Raspberry Pi 4, from November 2023.  I just ran an update (for the first time) using "sudo ./update.sh".  Everything appears have installed fine:

```
HEAD is now at bea3801 If geolocation API returns an error, print the whole thing
wire-pod.service created, building chipper with VOSK STT service...
Syncing...
wire-pod is now running with the updated code!

Updated successfully!
```

Now when I navigate to http://escapepod:8080/ I get the following message box:

`This webroot does not match with the wire-pod binary.  Some functionality will be broken.  There was either an error during the last update, or you did not precisely follow the update guide.`

I haven't been able to find any update guide anywhere.  The bot appears to work the same as before.   I see a new "Version Info" icon on the website, when I click it, it's stuck on "Checking for updates..." perpetually.

kercre123, 2024-05-05:
https://github.com/kercre123/wire-pod/wiki/Things-to-Know#updating-wire-pod

choozuradventure, 2024-05-05:
> https://github.com/kercre123/wire-pod/wiki/Things-to-Know#updating-wire-pod

I appreciate the link, admittedly I completely missed looking under Things to Know.  My original installation was from source.  I followed the instructions as stated:

```
sudo systemctl stop wire-pod
cd ~/wire-pod
sudo ./update.sh
sudo ./setup.sh daemon-enable
sudo systemctl start wire-pod
```

However I still get the alert on the web interface.

choozuradventure, 2024-05-05:
> https://github.com/kercre123/wire-pod/wiki/Things-to-Know#updating-wire-pod

I admittedly missed this section.  I appreciate your help.  This indeed made the alert go away.

closed 2024-05-05 by choozuradventure