kercre123/wire-pod #331: [HELP] Switching

jamesfawcett, 2024-05-16:
Hello, how can I cahnge language model from VOSK to Pico? 

I changed "vosk" to "pico" in apiConfig.json but that didn't work. 

I have the message:

`armv7l (32-bit) WARN: The Coqui and VOSK bindings are broken for this platform at the moment, so please choose Picovoice when the script asks. wire-pod is designed for 64-bit systems.`

My wire-pod is running `armv7l`

kercre123, 2024-05-18:
Could you describe exactly what the hardware is? I created .deb files for armhf, arm64, and amd64. Here is the one for armhf: [https://github.com/kercre123/WirePod/releases/download/v1.1.21/wirepod_armhf-v1.1.21.deb](https://github.com/kercre123/WirePod/releases/download/v1.1.21/wirepod_armhf-v1.1.21.deb)

jamesfawcett, 2024-05-21:
Thanks for your reply @kercre123 

I installed by compiling from source after pulling from git, and have been doing updates that way too.

I've removed the files and swapped to the image you suggested above, but now I no longer have the update.sh or setup.sh scripts, I can see files in /etc/wire-pod but not the same layout. 

With he release it's also appearing on ports 8080 and 80, but I needed port 80 for something else.

I don't have the warning about Vosk being unusuable now using release.

closed 2026-11-15 by kercre123