kercre123/wire-pod #333: [HELP] Lost SSH Key

evanpurl, 2024-05-26:
I booted up my OSKR bot to put into wire-pod, only to realize that I lost my ssh key for the bot when my old laptop died. What's the best practice to getting a new one? (I tried downloading the bot's logs when connecting to the bot via the setup website, didn't show a key.)

**What are you running wire-pod on?**
  - [OS]: Ubuntu
  - [Hardware]: Desktop

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
dev/OSKR

**What steps have you tried?**

Tried setting up wire-pod using the mode for OSKR, tried to set up bot but it asks for SSH key that I lost. Tried to download logs using web setup site, didn't have a key.

evanpurl, 2024-05-26:
I can't see the OS or Hardware answers, so here they are just in case:
OS: Ubuntu
Hardware: intel desktop, 6 or so years old

kercre123, 2024-05-28:
What firmware is the robot running? You can check by going into the CCIS screen (place vector on charger, double press button, lift the lift up then down)

The key should be contained inside of the log. IIRC you should just be able to extract it and it should be at ./data/ssh/id_rsa-Vector-#### or something like that

evanpurl, 2024-05-28:
> What firmware is the robot running? You can check by going into the CCIS screen (place vector on charger, double press button, lift the lift up then down)
>
>
>
> The key should be contained inside of [...]

Version 2.0.1.6076ep

I have tried to connect to the bot, and download the logs, but it was never contained anywhere in the bot's downloaded files. Not sure if I need to throw an older version of an OS on vector or anything to grab the key. I have also tried downloading the logs while in recovery mode.

kercre123, 2024-05-28:
That is a prod firmware, meaning you have a non-unlocked bot.

closed 2024-05-29 by evanpurl