kercre123/wire-pod #335: [HELP] Make input for custom host instead of escapepod.local

kenny-nt, 2024-05-27:
**Describe the issue:**
"Error logging in...." The bot seem can not communicate with the wire-pod instancce
**What are you running wire-pod on?**
  - [OS]: (ex. Windows, Debian)
  - [Hardware]: (ex. Raspberry Pi)

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
A product bot
**What steps have you tried?**
As I know the host  escapepod.local must be resolved but we can not edit or enter host manually?
Such as this step after clicking the Activate button if escapepod.local can not be resolve (no such host) > show input for entering custom host or ip???

orion14794, 2024-05-27:
> **Describe the issue:** "Error logging in...." The bot seem can not communicate with the wire-pod instancce **What are you running wire-pod on?**
>
>     * [OS]: (ex. Windows, Debian)
>
>     * [...]

I'm having same issues with my setup. I did the ./update.sh command line. My bot ip is example: 10.0.0.74, the error states ip 10.0.0.8:8080 is not reachable. Everything was fine before doing the update command. Runnng EscapePod on a Raspberry Pi 4, 4GB ram, Vector Gen 1, Firmware 2.1, Regular.

kercre123, 2024-05-27:
> **Describe the issue:**
> "Error logging in...." The bot seem can not communicate with the wire-pod instancce
> **What are you running wire-pod on?**
>   - [OS]: (ex. Windows, Debian)
>   - [Hardware]: (ex. [...]

The Web Bluetooth site tells Vector to activate. It does not communicate with the wire-pod instance itself. There is some other problem.

kenny-nt, 2024-05-28:
> The Web Bluetooth site tells Vector to activate. It does not communicate with the wire-pod instance itself. There is some other problem.

I read the log. 
`escapepod.local no such host`
Installed wire-pod on Raspberry Pi 4 and I set a custom routing escapepod.local to RP4's IP. And I can access to escapepod.local:8080, but still can not activate my Vector.
The point is why activation process always need host escapepod.local??? Why can't we change to some custom host or just an ip of wire-pod instance?

kercre123, 2024-05-28:
The URL is embedded in Vector's firmware and can't be changed. It is how DDL implemented escape pod functionality, which wire-pod is taking advantage of.

kenny-nt, 2024-05-29:
> The URL is embedded in Vector's firmware and can't be changed. It is how DDL implemented escape pod functionality, which wire-pod is taking advantage of.

So if DDL server is ON, how could Vector connect to it if it resolve local address escapepod.local????

kercre123, 2024-05-29:
It is a special property of the new "ep" firmware you are supposed to put on him. This was created and signed by DDL for escape pod. Normal firmware has public URLs embedded.

kenny-nt, 2024-05-29:
> It is a special property of the new "ep" firmware you are supposed to put on him. This was created and signed by DDL for escape pod. Normal firmware has public URLs embedded.

So there is no way to create a public wire-pod server right?

kercre123, 2024-06-04:
It is possible, though you'd have to run an mDNS broadcast app or set a DNS entry on your network to make sure `escapepod.local` resolves to the instance.

closed 2024-06-04 by kercre123