kercre123/wire-pod #340: [Feature Request] Support for ollama (local LLMs)

JMcrafter26, 2024-06-09:
Hey,
I saw there is support for OpenAI, so the structure for the request and response is there. This means it would be easy to implement an option for [Ollama](https://github.com/ollama/ollama) (Local LLMs, same api as OpenAI).

Please consider adding this, this would be very useful!

TypicalZedF, 2024-06-15:
I agree! I have Ollama installed on my Linux Server. I would enjoy a way to take Ollama output and run it through wire-pod. It's free and runs on your hardware!

ocueye2, 2024-06-15:
I dont know exactly how this works so take with a grain of salt but it should be as easy as adding a defined ip addr wherever the gpt api is defined

JMcrafter26, 2024-06-15:
Ollama uses the same api structure as OpenAI so it should be easy. You just need to be able to edit the endpoint and the model you want to use

LordJuice1, 2024-06-20:
This seems pretty similar to this request. Which I'm currently trying myself. I'm not sure where/how to make it where you can change it on the escapepod page and not in the script

https://github.com/kercre123/WirePod/issues/16#issue-2354622490

kercre123, 2024-06-23:
A "Custom" option for Knowledge Graph has been added, which lets you enter a custom endpoint.

closed 2024-06-23 by kercre123

LordJuice1, 2024-06-23:
Thanks Wire!

thefunkygibbon, 2024-09-20:
does it work?  I've put in my ollama server details
api key = ollama
endpoint = http://192.168.0.2:11434/api
model = (tried a bunch but for arguments sake using) llama3:latest (or is it just llama3  ?  I've tried both and both don't seem to work. 

every time i try to use vector it just tells me cannot connect to the LLM
`
2024.09.20 17:52:15: LLM error: error, status code: 404, message: json: cannot unmarshal number into Go value of type openai.ErrorResponse
2024.09.20 17:52:15: Intent matched: intent_system_unmatched, transcribed text: 'what's the answer to life the universe and everything', device: 00601532
`

JMcrafter26, 2024-09-21:
It worked in the past, but I tested it today (with a different setup) and I also can't get it to work

LordJuice1, 2024-09-23:
I'll admit, I'm not running the latest version. But, this is how my config page looks like. I'm also using OpenWebUI for my Ollama/llm needs.
`API Key- sk-167c2*****4cd4a************(this is a 35 character API key)`
`API Endpoint- http://xxx.xxx.xxx.xxx:11434/v1`
`Model name- llama3`

JMcrafter26, 2024-09-23:
I tried again and it worked.
My settings:
KEY: ollama
URL: http://192.168.2.146:11434/v1
MODEL: llama3.1:latest

uyuutityiugyu, 2025-04-16:
How did you find your url and will it be the same for mine