kercre123/wire-pod #356: [Feature Request] Support edge TTS, allowing Vector to answer in languages other than English

cantoblanco, 2024-07-03:
Hi, I integrated Openai into wire-pod and found that Vector can't read out replies in Chinese even if openai replies in Chinese. It seems that Vector‘s tts that can't read out languages other than English

So I suppose, considering that for example Edge-tts (https://github.com/rany2/edge-tts) can be used for free, could it be possible to integrate it into wire-pod so that Vector can reply in other languages?

kercre123, 2024-07-03:
In the latest commit/release of wire-pod, when OpenAI is selected, the inbuilt TTS is overwritten and an OpenAI voice is used instead, which supports many languages. It only works in intent-graph mode(i.e. "Hey Vector, what is the meaning of life?", and not "Hey Vector, I have a question,..."

cantoblanco, 2024-07-03:
> In the latest commit/release of wire-pod, when OpenAI is selected, the inbuilt TTS is overwritten and an OpenAI voice is used instead, which supports many languages. It only works in intent-graph mode(i.e. "Hey Vector, what is the meaning of life?", and not "Hey Vector, I have a question,..."

Yes I understand that, but custom models don't seem to be supported