kercre123/wire-pod #362: Request: Way to stop LLM response

MicaelJarniac, 2024-07-08:
Sometimes, I'm trying to tell Vector to do something, and the STT gets it wrong. It then doesn't get matched to a predefined sentence and is sent to the LLM. The response from the LLM is sometimes quite long, resulting in having to wait quite a while for Vector to stop talking.

It'd be nice to have something, like a press of its button or a pet on the back, to stop it from talking.

Zomboy4313, 2024-07-08:
A workaround I have is to tell the LLM "If a question seems cut off or you do not understand it, answer with “.{{playAnimationWI||confused}}.” And nothing else. This should make the LLM only play the animation. I tried with just a full stop but WirePod would crash for me

MicaelJarniac, 2024-08-01:
Closed by #368.

closed 2024-08-01 by MicaelJarniac