kercre123/wire-pod #363: [BUG] Saying "game" in any part of a sentence seems to default to Blackjack

MicaelJarniac, 2024-07-09:
**Describe the bug**
Trying to ask the LLM for information about a game seems to always get Vector to start playing Blackjack.

**To Reproduce**
Steps to reproduce the behavior:
1. Say "Hey Vector, what can you tell me about the Minecraft game?"
2. Vector starts Blackjack

**Expected behavior**
The prompt shouldn't match any known intents, thus forwarding it to the LLM.

**Screenshots**

**Device you are running wire-pod on (please complete the following information):**
 - OS: Ubuntu
 - Hardware: Dell Laptop

**Additional context**
```
2024.07.08 20:56:55: Intent matched: intent_play_blackjack, transcribed text: 'what can you talk about the minecraft game', device: 00706f36
2024.07.08 20:57:07: Intent matched: intent_play_blackjack, transcribed text: 'what can you tell me about the mind curve game', device: 00706f36
```

MicaelJarniac, 2024-07-09:
A possible solution would be to have something similar to "I have a question", but for talking directly to the LLM, thus skipping intent matching.

Zomboy4313, 2024-08-01:
WirePod is intent matching since you said 'game', saying I have a question is the method to always send responses to the LLM

MicaelJarniac, 2024-08-01:
Solved by #368.

closed 2024-08-01 by MicaelJarniac