kercre123/wire-pod #366: [Feature request] key button for interact with Vector

matte78, 2024-07-25:
I have installed wirepod over an Arduino board. Sometimes I would interact with Vector using my mobile, but I cannot use the keyboard for doing this (like moving it).
Is it possible to show the equivalent buttons instead of using the keyboard?

Zomboy4313, 2024-07-25:
Mobile remote control is not a feature yet, you'll have to use some sort of external keyboard. Hopefully mobile controls come soon

matte78, 2024-07-25:
[image]

something like this, for the debian package, but that works when I load the GUI from my mobile, works fine, I've tried it with my phone.
What do you think if I create the PR for this patch? should be useful? (ok I'm a backend developer, so the test is ugly, but I can do a better GUI if you agree!)

matte78, 2024-07-25:
[image]