kercre123/wire-pod #369: [HELP] My vector still can't talk with me

wenfeng0402, 2024-07-30:
**Describe the issue:**
I tried 5 times to install wirepod on my raspberrypi4,but my vector still cannot talk with me,I tried "vosk"
**What are you running wire-pod on?**
  - [OS]: (Ubuntu server)
  - [Hardware]: (Raspberry Pi4B)

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
100%prod,I bought at 2020
**What steps have you tried?**
I reinstalled not only the system but also wirepod,
BUT!!MY ROBOT STILL CANNOT TALK TO ME

kercre123, 2024-07-30:
Please provide more information. Be as detailed as you can. What instructions did you use to install wire-pod? Does Vector respond to commands? What do you mean by "CANNOT TALK TO ME"? Vector's main feature is not to talk. He will respond to most commands with emotion.

wenfeng0402, 2024-07-30:
That is, vector didn't respond to my orders, just like there were no members before.

wenfeng0402, 2024-07-30:
I connect my raspberrypi by wifi

wenfeng0402, 2024-07-30:
Except for not being able to respond to my request, all other settings are ok.

wenfeng0402, 2024-07-31:
Okay fix it,internet problem

Zomboy4313, 2024-08-01:
If the issue is fixed please close it to signal that

closed 2024-08-01 by kercre123