kercre123/wire-pod #375: [HELP] upload the bot's SSH key here,

Jeromaey, 2024-08-08:
**Describe the issue:**

**What are you running wire-pod on?**
  - [OS]: (ex. Windows, Debian)
  - [Hardware]: (ex. Raspberry Pi)

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?**
yes
**What steps have you tried?**
I don't know how to get,ssh key. I can't find that file !that's last step ,please help me

Jeromaey, 2024-08-08:
somebody help me ,please!how can I get the  ssh key

IAmDazen, 2024-08-08:
if you dont have an OSKR bot then you're using the wrong setup

IAmDazen, 2024-08-08:
also putting yes to "Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?" doesnt help at all

frien137, 2024-08-08:
Your bot is likely a production regular vector. People with Dev bots paid much more. You must follow the correct instructions for your bot. You must give meaningful answers to get help. You need to answer all of the questions.