kercre123/wire-pod #379: [BUG] SDK doesn't seem to work on newer versions

Zomboy4313, 2024-08-21:
**Describe the bug**

I'm not sure what happened but when i tried to use the sdk i was getting errors, reinstalling the sdk keeps failing to download the certificates by the looks of it. I'm using the SDK on a Mac and WirePod is installed on a Windows Machine

**To Reproduce**
Steps to reproduce the behavior:

Install SDK

**Expected behavior**

It should not fail 

**Screenshots**
[image]

**Device you are running wire-pod on (please complete the following information):**
 - OS: Windows
 - Hardware: AMD Ryzen 7 5700U Mini PC

kercre123, 2024-08-21:
This means you didn't clear user data before setting up wire-pod or there was some problem during authentication. This is the "weird behavior" that was described.

Zomboy4313, 2024-08-21:
Ah ok, that makes sense as I just reinstalled Windows and thus WirePod and didn't realise this is the weird behaviour that happens

Zomboy4313, 2024-08-21:
Can confirm this fixed my issue, thanks and sorry for not trying this out beforehand

closed 2024-08-21 by Zomboy4313