kercre123/wire-pod #393: [BUG]

Alexanderzawydiwski, 2024-09-27:
I’m having a problem with Face manager it says there’s no faces found and to set up another one, but vector still memorizes me and multiple other faces I tried rebooting the server and the robot but it hasn’t fixed it. Wire pod is running on a raspberry pi 4B any ideas ideas on a temporary fix or do I need to reset the robot

[image]

Sta2s, 2024-10-31:
Having exactly same issue right after I've tried to rename one of the registered faces. I could add new faces as well and if Vector recognizes me and I'm trying to re-register myself it says "you're not X you're Y" however the renamed user did register again the second time and now when Vector sees the glitched person he's just saying something like "E" (but that's likely an outcome of trying to say something like NULL or 0xF)
Will be looking into debugging my setup, would appreciate any advise on how to debug it further.
Other functionality continue working as expected, at least what I've manage to test so far.

Switch-modder, 2024-12-08:
What I had to do in this case is clear my bot's user data.