kercre123/wire-pod #403: [HELP] I always get error 203 when I run ota-start

whoobee, 2024-11-10:
**Describe the issue:**

**What are you running wire-pod on?**
  - [OS]: Windows / Debian
  - [Hardware]: Windows PC / Raspberry-Pi

I had it a long time ago set-up as dev/OSKR but then I switched back to a production build.
To be honest, I did this a long time ago, and I don't remember what were the last steps I did on the bot.
I can see that my bot has the following FW: v0.9.0-087cafe_os0.9.0-4ee03ec-202010021627
[image]

When I've tried all the things below, I always cleared the cache from the robot and also put it into development mode (kept the button on the back pressed for 15s)

- I've tried setting up the wire-bot on both windows and debian (raspberry-pi)
- I've set-up the wire-bot as both the OSKR and Production bot configuration
- On the automatic set-up I am able to connect to WiFi but when I start the update I get the exclamation mark on the vector screen like the OTA update failed
- I've also done the steps to update the bot from a local http server (configured on my windows host machine) and this is how I see that I always get the 203 error code
- When I've started the OTA, I used the local IP (not localhost of course) and I can see that vector is trying to connect for about 5 minutes and after that I get the 203 error (so it is not immediately)
[image]
- Once connected to WiFi, I can ping the Vector from my windows PC, so I know the connection is available
- I also tried the internal AP of Vector, and connected my PC directly to vector's AP and tried this way, but again I get the 203 error
[image]
- as a different step I also tried the ddl vector-ble application, but again the same result, when I try to do the OTA update I get 203 :(

Am I missing a step here, has anyone have experienced something similar?
Any help is very appreciated

Switch-modder, 2024-12-08:
So what you are trying to use is your bot's unlock image, this image only needs to ever be used once.
Run: ota-start http://ota.pvic.xyz/OSKR_EP_1.7.3_6016.ota
This will put a official OSKR ota on the bot.

whoobee, 2024-12-29:
> So what you are trying to use is your bot's unlock image, this image only needs to ever be used once. Run: ota-start http://ota.pvic.xyz/OSKR_EP_1.7.3_6016.ota This will put a official OSKR ota on the bot.

Thanks a million, it worked! 💪 
Feels so good to have it back alive! 😀