kercre123/wire-pod #417: [BUG] Vosk STT can't be changed on Android

iam-best, 2025-01-06:
Description:
Basically when you want to download a better Vosk Model, for example the 3 GB one you can't import it on Android!

How to reproduce?
1. Install the Wirepod .APK!
2. Go to the Wirepod Folder in Android Data
3. The only Folder that exists is an Empty Files Folders
4. You can't import VOSK or at least not in the same way as on PC

Expected Behavior:
The Folder should be the same as on PC and you should be able to import the Vosk Model or change it in config, maybe let us download it via the Config Page.

Device and OS:
 - OS: Android 13
 - Samsung Galaxy Tab S6 Lite

Additional Context: If I'm mistaken please let me know how to get the better VOSK STT, thanks!

phatin22, 2025-01-06:
From my testing, Wire-Pod on android seems to store it's main files in /data/user/0/com.kercre123.wirepod/files and not in /sdcard/android/data/com.kercre123.wirepod/files

From what I've tried, you can't access /data/user/0 on an unrooted device (Device: Pixel 8, Android 15 [unrooted])
[image]

This is what the wirepod folder on /data/user/0 looks like (Device: Honor 8, Lineage OS 16 [Android 9] [Rooted])
[image]

iam-best, 2025-01-06:
Thanks for the response, by connecting your Phone/Tablet to PC you can access it but as I said there's just the Files Folder which is completely empty! (Also I wasn't on SD Card)

iam-best, 2025-01-06:
I will try again since you see the Files.

iam-best, 2025-01-06:
Checked again, still not working... The files and folders in the files folder just don't exist. How can I fix it?

phatin22, 2025-01-06:
> Thanks for the response, by connecting your Phone/Tablet to PC you can access it but as I said there's just the Files Folder which is completely empty! (Also I wasn't on SD Card)

Despite its name "/sdcard" is internal storage of an android device. I can confirm that nothing exists in /sdcard/Android/data/com.kercre123.wirepod contains nothing. 
[image]

phatin22, 2025-01-06:
> Checked again, still not working... The files and folders in the files folder just don't exist. How can I fix it?

My guess would be to root the Android device (Not Recommended) or swap to a different device like a pi or another computer.

iam-best, 2025-01-06:
I mean the Folder Files exist but it just doesn't have anything in it. And also everything else like Minecraft World Data work just fine which is why I think it's a bug.

phatin22, 2025-01-06:
> I mean the Folder Files exist but it just doesn't have anything in it. And also everything else like Minecraft World Data work just fine which is why I think it's a bug.

Wire-Pod doesn't keep anything necessary for its function (like the vosk voice model) in the Android/data folder. It's all stored separately in /data/user/0/  (not accessible on an unrooted phone)