kercre123/wire-pod #424: [BUG]

damoebius, 2025-01-18:
**Describe the bug**
My vector can't speek. It still make sounds effects, but can't say anything.
Also, asking for it's age, make it busy and fail.

**To Reproduce**
Steps to reproduce the behavior:
1. Ask 'what time is it'
2. Display time but doesn't say it

**Device you are running wire-pod on (please complete the following information):**
 - OS: RaspberryPi OS
 - Hardware: Raspberry Pi 5

clarin-ebtio800090, 2025-01-18:
Hello, we tried to solve the issue.

This is what we did:

Modified the DoSayText function to ensure Vector speaks, and added error handling. Also added a check for age-related queries to prevent failures.

You can review changes in this commit: https://github.com/clarin-ebtio800090/kercre123-wire-pod-424/commit/31953d638fe097f1f9e8e80086524278a9f65870.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

bliteknight, 2025-01-18:
If you language is set to anything other than English, and voice related responses will fail. So check what language your Vector is set to.

damoebius, 2025-01-19:
It was English US and since today, i tried french FR but still no voice

iam-best, 2025-01-24:
You sure the Language is for both Vector and the Server set to English? This is because Vector doesn't have a TTS for other Languages besides English currently which makes him not being able to speak!