kercre123/wire-pod #425: [BUG]  Feature Request and Bug Report

iam-best, 2025-01-24:
Description:
Wirepod crashes when using the German STT and renaming it to en-US and applying it.

How to reproduce?
1. Go to %appdata%
2. Click on wirepod
3. Change the any other Language from Vosk to en-US
4. Say "Hey Vector" and Wirepod crashes

Expected Behavior:
Vector should still respond normally and work and this would be useful to have the Server in English but the STT in another Language.

Device:
 - OS: Windows 11
 - Hardware: Ryzen CPU

Additional context:
You can already do this by tricking Wirepod somehow but this is useful if you can just ask him in for example German and he responds still in English by telling ChatGPT to only respond in English, this is already possible using the ChatGPT Prompt Feature and the Server should only be the STT and Vector's Language shouldn't change just because I asked him in for example German.

bliteknight, 2025-01-25:
Can you include the logs when you do this or some more context. Did you change the folder reference while WirePod was running, and does it still crash after you restart?

This is just a guess, but it could be a Vosk error if it's expects a English model, but you provide a German one

iam-best, 2025-01-25:
It worked one time and also how do I restart Wirepod I always end it in the Task Manager and open it up again. The logs say nothing.