kercre123/wire-pod #428: [BUG]

AverWasTaken, 2025-02-05:
**Describe the bug**
A clear and concise description of what the bug is.

Vector will not respond to commands and shows the wifi error symbol. (The one with the !)

**To Reproduce**
Steps to reproduce the behavior:
Use wire-pod on Docker, and ask Vector a question.

**Expected behavior**
A clear and concise description of what you expected to happen.

Vector should respond to commands. Alexa works, just not anything "Hey Vector" related.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device you are running wire-pod on (please complete the following information):**
 - OS: Ubuntu
 - Hardware: Lenovo Thinkpad, Core i5.

**Additional context**
N/A

iam-best, 2025-02-07:
Does Wirepod crash?

Henry140501, 2025-07-13:
Gosh. I have the same problem. I'm on Windows. The wirepod is working fine in everything, except that every time I use the trigger word, wirepod disappears(or you can say crashes somehow). Then I'll have to restart it. Help!!!

iam-best, 2025-07-13:
What Vosk model are you using? Did you also configure your Api's properly?