kercre123/wire-pod #430: [BUG] Deepseek not working on Wirepod using Together API

iam-best, 2025-02-06:
**Describe the bug**
When using Deepseek in the Together API Wirepod will crash and Vector will not respond with the full sentence.

**Steps to reproduce the behavior:**
1. Go to Knowledge Graph
2. Choose Together API
3. Choose Deepseek as a Model
4. Say "Hey Vector" or press his Button and say something that his Brain doesn't understand without going to Together API.
5. See error

**Expected behavior**
It should just work.

**Device you are running wire-pod on (please complete the following information):**
 - OS: Windows 11
 - Hardware: Ryzen CPU

**Additional context**
I'm not the only one with this Problem as a YouTuber tried the same thing and it didn't work either.

I personally don't care about this Bug since I don't use Deepseek but I would try it and other might want to use it!

DamianS-eng, 2025-02-22:
Get the link of the YouTube video you saw.
Or at least the guy's channel name.

iam-best, 2025-02-22:
https://youtube.com/shorts/D160x0dxHIk?si=kapFFv8ipBLTS4hU

For me Vector said what Deepseek thinks and then stops while talking... I think for some people Wirepod crashed!

iam-best, 2025-02-22:
@DamianS-eng