kercre123/wire-pod #441: [HELP] Running Vector with magnetic base upside down on a steel shelf

tinkrmind, 2025-04-20:
Thanks everyone in this community for making WirePod accessible. 

I revived my vector using WirePod yesterday, and today I installed [magnets on its base](https://photos.app.goo.gl/gfsaH4HAqEsMcAnk7), in place of the weights. It took 24 magnets, but the little fella sticks well enough(at least well enough for testing).

However, he's all the grumpy, just trying to get right side up, and throwing a fit([video clip](https://photos.app.goo.gl/e8HD9v7LHVPuJpj39)). I expected this, and was hoping I could flip the IMU to fix the issue, but turning the whole main screen board upside down didn't work. The way I see it, I have three options:
1) Find some way to change the firmware - Doesn't seem possible
2) Operate it remotely using WirePod SDK - Seems kinda sad, like I'm taking away its autonomy
3) Man in the middle the IMU changing the data to look right side to Vector - Seems hella complicated and with the amount of internal compute LSM6DSO has, might not work anyway

Wondering if you have any advice for me on how to have the little dude running upside down!

Switch-modder, 2025-04-20:
The IMU is located in the bodyboard. With a OSKR or DEV robot you can likely make it accept the flipped data.
Try asking around in the server here
https://discord.gg/vector-friends-527874754342944770

kercre123, 2025-04-20:
He has a BMI160 IMU on the headboard. As Switch-modder said, with an OSKR or DEV robot, the firmware can probably be modified to work fine with it. With a regular production robot, I don't see any way forward.

This is also very out-of-scope for wire-pod. Go to the Discord server Switch-modder linked.

closed 2025-04-20 by kercre123