kercre123/wire-pod #447: [HELP] After holding backpack button and using updater - progress bar never moves?

gleep52, 2025-06-01:
**Describe the issue:**
Put Vector in firmware update mode - ddl.io/v shows up - double click to pair him - says it's updating and shows an empty progress bar.  Vector's display shows the cloud with rotating arrows indicating something is processing.  I have turned on experimental web features in Edge and Chrome - same results.  Empty bar with no progress indication... I've let it run for over 30 minutes and nothing happens.  

**What are you running wire-pod on?**
- I setup a proxmox VM with Ubuntu 24 on it but haven't gotten to that step yet!  
  - [Hardware]: (ex. Raspberry Pi) Proxmox host is AMD Zen2 CPUs - VM has 8gb dedicated so far.

**Are you setting up a prod (regular) bot or a dev/OSKR (unlocked) bot?** This is a brand new vector2  from anki - not a dev unit - but I DID pay for the subscription for a year as well.

**What steps have you tried?**
Explained above already - just not sure how long the update might take - and what slows it down?  I've got a 5gb fiber up/down and no traffic right now.  Is this hosted on your servers, DDL's servers, and what's the address - I see nothing blocked on my firewall, but perhaps the website it's trying to obtain data from is down or not responding?

bliteknight, 2025-06-02:
When flashing the appropriate firmware, sometimes it fails when coming from the source baked into WirePod
Try flashing from here: https://vector.techshop82.com
If you still get the exclamation mark in the middle of the cloud, try moving your Vector closer to your Wifi router.
The fastest it should take to flash the firmware is about 3 mins, the longest I've seen is 45mins

gleep52, 2025-06-03:
So does Vector download directly from Anki.bot or does it use bluetooth to download from my browser like a proxy?  I'm a network guy/sysadmin/cybersec guy now - and honestly if it's actually downloading data - it's nearly invisible in bandwidth from his device directly.  I was able to get it to update ONCE and it took over an hour using a macbook pro without power saving enabled.  

Sadly - I built a new dedicated wirepod ubuntu box and now I seem to have found myself needing to reset the poor guy again and am facing the same issue.  The site you listed above seems to give you a drop down option to pull from but it does not seem to be any faster or slower - the green bar simply doesn't progress at all with either site.  It's been 5 minutes of a connection (no bluetooth disconnects yet) and no green bar has appeared.  I can't move any closer to the access point - it's a dedicated 2.4ghz SSID for Vector about 6 feet away line of sight access.  

Just curious if this is an anki problem or if it's my network related possibly.